LeandroSQ/android-ble-made-easy #1: react to events emitted from the ble device

Ditscheridou, 2021-04-20:
Hi! 

i just stumblep upon your ble lib. Seems very well done ,congratz! When i looked through the code i asked myself if the lib can already react to events that are emmitted from the ble device?

LeandroSQ, 2021-04-21:
Hello there!
First of all, thanks for the interest on the project, I appreciate it

When you say 'react to events' you mean to constantly read a characteristic value and whenever it changes trigger a callback?
Or did I missunderstood the sentence?

Ditscheridou, 2021-04-21:
Yeah that’s what I meant. I have the requirement for that and possible can implement it (not sure tho), but I think that would be nice to have as a standard feature in the library

LeandroSQ, 2021-04-21:
Yeah, it would be a nice feature indeed!
Unfortunately, I don't have any BLE devices at hands now, for at least a couple of weeks
(And implementing such feature require testing on actual devices)

But until it isn't embed on the library itself, you could try using something like this:
```kotlin
private fun startReadings(device: BluetoothConnection) {
      val pollingRate = 1000L// Define the interval between the readings, in ms
      var lastValue: String? = null
      val characteristic = "0.0.0.0.0"// Define the characteristic to keep track of
      
      GlobalScope.launch {
	      // Reads the characteristic from the device
	      val currentValue = device.read(characteristic, Charsets.UTF_8)
	      // Check if it has changed
	      if (lastValue != currentValue) {
		      // TODO: Call a callback or do some action when the value changes
		      
		      // Update the lastValue to reflect the changes
		      lastValue = currentValue
	      }			
	      
	      // Waits until the next reading
	      delay(pollingRate)
      }
}
```

After checking permissions, starting the BLE instance
And connecting to a device you want, just call the method passing a established bluetooth connection (Which can be acquired using `scanFor`, `scanForAsync`, `scan` and `scanAsync`) and you should be ready to go

Needless to say that you can tweak the vals on the method, polling rate and characteristic to fit your needs.

Ditscheridou, 2021-04-23:
Thanks for the sample! I will try it out and improve if necessary, if its working well enough i will make a pr!

LeandroSQ, 2021-11-05:
At the moment working on this at branch [feature/demo](https://github.com/LeandroSQ/android-ble-made-easy/tree/feature/demo)

LeandroSQ, 2021-11-12:
Added support for BLE NOTIFY property and legacy observe, that implements a similar concept as the presented above, on the version 1.5.2.

closed 2021-11-12 by LeandroSQ