LeandroSQ/android-ble-made-easy #2: Wrong dependencies ?

leogala, 2021-05-05:
Hi, 
very nice work, just a small issue I had, I might be wrong but I think that the dependencies in the documentation are wrong, adding a "v" seems solves, for example for  version 1.4.0 : 

implementation 'com.github.LeandroSQ:android-ble-made-easy:v1.4.0'

LeandroSQ, 2021-05-05:
hello there!

Thanks for your interest on the project, I appreciate it!

You are right, the readme is incorrect
On the jitpack page it lista the versions with the prefix 'v'

If you want to, you can create a Pull request fixing it, just mention this issue and I will approve it

If you don't want to, no problem, I will commit the changes myself

Thanks

LeandroSQ, 2021-05-10:
Fixed!

closed 2021-05-10 by LeandroSQ