LeandroSQ/android-ble-made-easy #3: No device scanned

D10NGYANG, 2021-06-04:
2021-06-04 10:50:14.666 12002-12002/quevedo.soares.leandro.androideasyble D/MainActivity: Setting bluetooth manager up...
2021-06-04 10:50:14.684 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Checking App bluetooth permissions...
2021-06-04 10:50:14.684 12002-12002/quevedo.soares.leandro.androideasyble I/HwPartMagicWindowFactory: add android.common.HwPartMagicWindowFactoryImpl to memory.
2021-06-04 10:50:14.684 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: All permissions granted!
2021-06-04 10:50:14.684 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Checking bluetooth adapter state...
2021-06-04 10:50:14.685 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Checking location services state...
2021-06-04 10:50:14.691 12002-12002/quevedo.soares.leandro.androideasyble I/HwPartPowerOfficeFactory: add HwPartPowerOfficeFactoryImpl to memory.
2021-06-04 10:50:14.697 12002-12002/quevedo.soares.leandro.androideasyble D/InputEventReceiver: dispatchInputInterval 1000000
2021-06-04 10:50:14.714 12002-12053/quevedo.soares.leandro.androideasyble D/HiTouch_PressGestureDetector: onAttached, package=quevedo.soares.leandro.androideasyble, windowType=1, mHiTouchRestricted=false
2021-06-04 10:50:14.721 12002-12047/quevedo.soares.leandro.androideasyble I/FrameFlow: FrameFlowInit app is not supported by frameflow solution
2021-06-04 10:50:14.721 12002-12047/quevedo.soares.leandro.androideasyble I/iGraphics: [0020080c] pn: quevedo.soares.leandro.androideasyble, p: 12002
2021-06-04 10:50:14.721 12002-12047/quevedo.soares.leandro.androideasyble I/iGraphics: [0030080c] no spt app: quevedo.soares.leandro.androideasyble
2021-06-04 10:50:14.729 12002-12047/quevedo.soares.leandro.androideasyble D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-06-04 10:50:14.732 12002-12047/quevedo.soares.leandro.androideasyble W/Gralloc3: mapper 3.x is not supported
2021-06-04 10:50:14.742 12002-12002/quevedo.soares.leandro.androideasyble I/HwViewRootImpl: removeInvalidNode jank list is null
2021-06-04 10:50:14.745 12002-12002/quevedo.soares.leandro.androideasyble I/RmeSchedManager: init Rme, version is: v1.0
2021-06-04 10:50:14.745 12002-12047/quevedo.soares.leandro.androideasyble D/OpenGLRenderer: disableOutlineDraw is true
2021-06-04 10:50:14.761 12002-12002/quevedo.soares.leandro.androideasyble D/DecorView: showOrHideHighlightView: hasFocus=true; winMode=1; isMrgNull=true
2021-06-04 10:50:14.761 12002-12002/quevedo.soares.leandro.androideasyble W/InputMethodManager: startInputReason = 1
2021-06-04 10:50:14.764 12002-12002/quevedo.soares.leandro.androideasyble W/InputMethodManager: startInputReason = 5
2021-06-04 10:50:14.777 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Starting scan...
2021-06-04 10:50:14.777 12002-12050/quevedo.soares.leandro.androideasyble I/BluetoothAdapter: isLeEnabled(): ON
2021-06-04 10:50:14.779 12002-12046/quevedo.soares.leandro.androideasyble D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2021-06-04 10:50:14.782 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Scan timeout reached!
2021-06-04 10:50:19.523 12002-12002/quevedo.soares.leandro.androideasyble I/AwareBitmapCacher: init lrucache size: 2097152 pid=12002
2021-06-04 10:50:24.786 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Stopping scan...
2021-06-04 10:50:24.787 12002-12050/quevedo.soares.leandro.androideasyble I/BluetoothAdapter: isLeEnabled(): ON
2021-06-04 10:50:24.790 12002-12050/quevedo.soares.leandro.androideasyble D/BluetoothMadeEasy: Scan finished! 0 devices found!

### My Phone
MODEL: HUAWEI YAL-AL00
SYSTEM: Android 10 API 29

LeandroSQ, 2021-06-04:
Hello, 

First of all, thanks for the interest on the library, I appreciate it!

What types of devices are your trying to scan?

LeandroSQ, 2021-06-14:
Closing the Issue for inactivity

closed 2021-06-14 by LeandroSQ

ChristoforosKor, 2022-08-26:
Hello. I am having the same issue.
I have raised the timeout to 50000.
My device is not discoverd. I can discover it with third party app nrf Connect which show the following for my BLE device:
Device type: LE only
Advertising type: Legacy
Flags: GeneralDiscoverable, BrErdNotSupported
Complete Local Name: Weight

LeandroSQ, 2022-08-26:
Hello there! @ChristoforosKor 
Thanks for sharing your issue!

Could you provide a code sample?

ChristoforosKor, 2022-08-27:
Hello,
This is the code I used ( I am sorry for the format, I didn't managed to make editor show it correctly)

`class ScanBagCodeActivity : AppBaseActivity(), DialogInterface.OnKeyListener {
    override fun onCreate(savedInstanceState : Bundle? ) {

super.onCreate(savedInstanceState)
        var ble = BLE(this)
        GlobalScope.launch {
            val granted = ble.verifyPermissions(
                rationaleRequestCallback = { next -> Log.i(TAG, "Get Permissions")
                    next()
                }
            )
            if (granted) {
                verifyBLEAdapter(ble)
            } else {
                Log.i(TAG, "Permissions not ok")
            }
        }

    }

    private fun verifyBLEAdapter(ble: BLE) {
        GlobalScope.launch {
            if (ble.verifyBluetoothAdapterState()) {
                turnOnLocationServices(ble)
            } else {
                Log.w(TAG, "Bluetooth is closed")
            }
        }
    }

    private fun turnOnLocationServices(ble: BLE) {
        GlobalScope.launch {
            if (ble.verifyLocationState()) {
                scanForDevices(ble)
            } else {
                Log.w(TAG, "Bluetooth is closed")
            }
        }
    }

    private fun scanForDevices(ble: BLE) {
        GlobalScope.launch {
            try {

                val devices = ble.scan(duration = 10000)
                Log.d(TAG, "---------------------------------->" + devices.size)
                for (device in devices) {
                    Log.d(
                        TAG,
                        "---------------------------------->" + device.name + "-" + device.advertisingId + "-" + device.macAddress + "-" + device.macAddress
                    )
                }
            } catch (e: Exception) {

                Log.d(TAG, "---------------------------------->" + e.message)
            } catch (e: ScanFailureException) {

                Log.d(TAG, "---------------------------------->" + e.message)
            }
        }
    }

}`

At code instruction
    Log.d(TAG, "---------------------------------->" + devices.size)

I always get 0 as you can see from the output excerpt below (last line).

2022-08-27 08:29:01.181 5779-5779/com.reciget.reciget V/InputMethodManager: onWindowFocus: null softInputMode=32 first=true flags=#81810100
2022-08-27 08:29:01.234 5779-8852/com.reciget.reciget D/FA: Connected to remote service
2022-08-27 08:29:01.235 5779-8852/com.reciget.reciget V/FA: Processing queued up service tasks: 1
2022-08-27 08:29:01.242 5779-5852/com.reciget.reciget D/OpenGLRenderer: ~CanvasContext() 0x7a2321d880
2022-08-27 08:29:01.552 5779-8886/com.reciget.reciget D/BluetoothLeScanner: startScan
2022-08-27 08:29:01.566 5779-8886/com.reciget.reciget D/BluetoothAdapter: STATE_ON
2022-08-27 08:29:01.609 5779-5792/com.reciget.reciget D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=7
2022-08-27 08:29:01.609 5779-5792/com.reciget.reciget D/BluetoothLeScanner: mClientIf=0
2022-08-27 08:29:01.628 5779-8886/com.reciget.reciget I/BluetoothLeScanner: startRegisteration: mLeScanClients={quevedo.soares.leandro.androideasyble.BLE$setupScanCallback$1@b5a6dd9=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@1cf139e}
2022-08-27 08:29:06.330 5779-8852/com.reciget.reciget V/FA: Inactivity, disconnecting from the service
2022-08-27 08:29:11.631 5779-8886/com.reciget.reciget D/BluetoothLeScanner: stopScan
2022-08-27 08:29:11.647 5779-8886/com.reciget.reciget D/BluetoothAdapter: STATE_ON
2022-08-27 08:29:11.648 5779-8886/com.reciget.reciget I/BluetoothLeScanner: startRegisteration: mLeScanClients={quevedo.soares.leandro.androideasyble.BLE$setupScanCallback$1@b5a6dd9=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@1cf139e} ,callback=quevedo.soares.leandro.androideasyble.BLE$setupScanCallback$1@b5a6dd9
2022-08-27 08:29:16.052 5779-8885/com.reciget.reciget D/Weight-APP:  **---------------------------------->0**