LeandroSQ/android-ble-made-easy #4: Not able to scan And print

iAmAndroidDeveloper, 2021-06-12:
@LeandroSQ  after integration of your library i am not able to scan and get BLE devices. I checked increase time out as well

i am creating demo for print 58mm BLE printer  i find name and mac address from other app but i dont what i need to pass as "characteristic" for write to printer?

i create receipt format but dont know hot to send this to BLE printer and after compete print disconnect with printer 

can you please help me for that

Thanks.

LeandroSQ, 2021-06-14:
Hello there, first of all thank you for the interest in the project, I appreciate it!

After reading your issue, 
I believe that your printer uses classic bluetooth to pair and receive print commands. Therefore you will not be able to scan, connect and neither communicate with it.

The library is tailored around the poorly functional BLE on Android, but unfortunatelly it wan't made with Classical bluetooth in mind. But there are plenty of libraries that would be useful for your application.

For further information [here is an article about the differences between BLE and Classical bluetooth](https://www.link-labs.com/blog/bluetooth-vs-bluetooth-low-energy#:~:text=When%20talking%20about%20Bluetooth%20Low,4.0's%20low%20power%20consumption.&text=Just%20like%20Bluetooth%2C%20BLE%20operates,when%20a%20connection%20is%20initiated.)

If you want to contribute to the project, by creating such support for Classical bluetooth, just create an PR and I will help as I can.

iAmAndroidDeveloper, 2021-06-15:
@LeandroSQ Thanks for your feedback

iAmAndroidDeveloper, 2021-06-15:
@LeandroSQ 
Now i am able to connect with BLE printer 

connection?.let {
                 // According to the 'active' boolean flag, send the information to the bluetooth device
                 val result = it.write("xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",   "Testing")
                 
                 I got true on result and connected to printer now how to test print receipt using connected printer
                 if (result) {
                   
                 } else {
                   
                 }
             }

LeandroSQ, 2021-06-15:
Do you know the model of the printer that you are using?

iAmAndroidDeveloper, 2021-06-15:
Please check I add images that I used

iAmAndroidDeveloper, 2021-06-15:
Some time able to printer and one more things that not able to print full string it's just print 1 line max rest of skipped

iAmAndroidDeveloper, 2021-06-15:
I think see this was the main problem that receipt not print bulk data. Below loops not able to wait for for process

gatt.services?.forEach { service ->
				service.characteristics.forEach { characteristic ->
					if (characteristic.uuid == characteristicUuid) {
						characteristic.value = message
						return gatt.writeCharacteristic(characteristic).also {
							if (!it) log("Could not write to device ${device.address}")
							this.finishOperation()
						}
					}
				}
			}

LeandroSQ, 2021-06-15:
I searched your printer model, and it seems that it makes uses the classic bluetooth

Being that clear, I do not recommend this Library to interface with your printer, since it wasn't made for that purpose.
Unfortunatelly, I do not posses any bluetooth printer to verify that, so I'll list some libraries that I believe could help you far better:
- https://github.com/mazenrashed/Printooth
- https://github.com/andrey-ushakov/esc_pos_bluetooth
- https://github.com/deysuman/Android-Bluetooth-printer

iAmAndroidDeveloper, 2021-06-15:
I Need library or code for only BLE device for Bluetooth(Pair and connect) i already done and its working fine, that's why i finding something that support all BLE printers and do print without any configs or issues

I used your library for only purpose of print and its already work when you give a chunks of data with used of using some millisecond's of delay (Like Handler or Somethings)

i think we can do something more on that so if possible then mail me(valanibhavesh50@gmail.com)

iAmAndroidDeveloper, 2021-06-27:
@LeandroSQ  after some days research finally got success to BLE search, and print text-image-Qr code without skipping any byte data

closed 2021-06-27 by iAmAndroidDeveloper