LeandroSQ/android-ble-made-easy #6: Getting started

Swarandeep1, 2021-11-04:
Hello,

This isn't really an issue as much as it is a request in getting help with using BLE made easy.  I'm some what new to app development and new to Kotlin.  

**Background**
I'm currently working on a project where I want to control LED lights via ble.  My first step is to do a simple toggle the lights on and off.  I'm using HM-10 BLE module to receive signals and an arduino uno to control the signals to the lights.  I've used the dabble app to test my HM-10 and arduino uno code and have successfully seen that the lights turn on.  Now I would like to build my own app to control the lights from my phone.

**What I'm Seeing**
I've imported ble-made-easy into android studio and made sure there are no errors.  When I test the app (via wireless debugging) on my device I only see main activity (screen title says "AndroidEasyBLE").  No list view or text views or buttons.

Do I need to do anything in the main activity to inflate the fragments and bind correctly?  Is there an example I can run out of the box, where I can scan, and connect to a BLE device?

Thank you very much!
Swarandeep Singh

LeandroSQ, 2021-11-04:
Hello there, thank you for writting the issue!

Well, I've realised with your issue that the example in master is not so good for new comers... Therefore I'll make some small changes to better demonstrate the library by just cloning the repository.

LeandroSQ, 2021-11-04:
What version of Android are you using?

Swarandeep1, 2021-11-04:
Hello,

It says Android Studio Arctic Fox | 2020 3.1 patch 2
runtime version 11.0.10...

Thanks,
Swarandeep Singh

LeandroSQ, 2021-11-05:
I've noticed that the library is not working properly on Android 11, since there were many updates regarding Bluetooth in this Android version.
I'm currently working on the android 11 support and it will be available ASAP.

LeandroSQ, 2021-11-05:
I've committed on the branch [feature/demo](https://github.com/LeandroSQ/android-ble-made-easy/tree/feature/demo) the fix for that.
Would you mind to try it and give feedback?

Swarandeep1, 2021-11-05:
After I follow steps 1 - 4, I try to run the app on my device and I get the following error:

Could not find com.github.LeandroSQ:android-ble-made-easy:null.
Required by:
    project :app
Search in build.gradle files

Once I removed the dependency, the example app worked great and I was able to connect to my HM-10 module.

When I go to incorporate the library into an app I'm currently building, I do the following:
1.) download/clone this git branch to a location on my desktop (https://github.com/LeandroSQ/android-ble-made-easy.git)
2a.) Copy the folder into my lib directory in Android Studio IDE for my project
2b.) add as library
3.) follow your steps 1 - 4 for installing  

Then Can I simply copy you adapter and view directories from the example into my app directories?  Or am I over simplifying it?

LeandroSQ, 2021-11-05:
Since you were cloning the repository and using the example App provided within it, you shouldn't have to add the implementation at the **build.gradle** files, that would be only if you are trying to embed the library into your application.

Since it worked with your peripheral with the latest changes, I'll be doing some more testing and a new version should be released soon. 
And then you only will need to follow the install steps at the README.md.

You can import the library on your project, but it would be so much pain for something that will be available for you by just adding one line to your gradle file.

I'm working on the library to release this ASAP. 
Thank you for contributing to the project, I really appreciate it.

LeandroSQ, 2021-11-08:
I've submitted a pull request to master.
Could you try now follow the steps at the README.md for importing the version v1.5.0?

Swarandeep1, 2021-11-08:
First Off, I want to say thank you for working with me.

I tried going through the new install steps and I'm getting sync & build failures.  I started a new project and just went through the Install steps, with one modification.

In my project I don't have an all projects attributes in my project build.gradle.  So, when I add:
_allprojects {
        repositories {
            maven { url 'https://jitpack.io' }
        }
    }_
I get the following error:
_A problem occurred evaluating root project 'BLEMadeEasy_tut3'.
> Build was configured to prefer settings repositories over project repositories but repository 'maven' was added by build file 'build.gradle'_

**Modification**: add _maven { url 'https://jitpack.io' }_ to buildscript -> repositories element.
Then I'm able to sync successfully.  I'm guessing its just the way I have my project and IDE configured.

Then add the dependency into my app build.gradle file and sync successfully.  However, when I got to run the project from there with nothing else done to the default "new project" code i.e. new project out of the box.  I get the following error:

_7: Task failed with an exception._
-----------
- What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find com.github.LeandroSQ:android-ble-made-easy:v1.5.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/com/github/LeandroSQ/android-ble-made-easy/v1.5.0/android-ble-made-easy-v1.5.0.pom
       - https://repo.maven.apache.org/maven2/com/github/LeandroSQ/android-ble-made-easy/v1.5.0/android-ble-made-easy-v1.5.0.pom
       - https://jcenter.bintray.com/com/github/LeandroSQ/android-ble-made-easy/v1.5.0/android-ble-made-easy-v1.5.0.pom
     Required by:
         project :app_

_...
Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/7.0.2/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 710ms
17 actionable tasks: 7 executed, 10 up-to-date_

Please let me know if I'm missing something blatantly obvious?

Thanks,
Swarandeep Singh

LeandroSQ, 2021-11-08:
Do you have a "settings.gradle" file at the root of your project?

If so... Ignore the step to add "maven { url 'https://jitpack.io' }" into your project build.gradle

Instead try opening the settings.gradle file and making sure that the start of the file looks like this:
```groovy
import org.gradle.api.initialization.resolve.RepositoriesMode

dependencyResolutionManagement {
    repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)

    repositories {
        google()
        mavenCentral()
        maven {
            url "https://www.jitpack.io"
        }
    }
}
```

Swarandeep1, 2021-11-09:
The deprecated gradle error has gone away. But I'm still getting this error:

`7: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find com.github.LeandroSQ:android-ble-made-easy:v1.5.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/com/github/LeandroSQ/android-ble-made-easy/v1.5.0/android-ble-made-easy-v1.5.0.pom
       - https://repo.maven.apache.org/maven2/com/github/LeandroSQ/android-ble-made-easy/v1.5.0/android-ble-made-easy-v1.5.0.pom
       - https://www.jitpack.io/com/github/LeandroSQ/android-ble-made-easy/v1.5.0/android-ble-made-easy-v1.5.0.pom
     Required by:
         project :app`

When I click on those location I get a 404 error.  Is it possible that there is something amiss in the GitHub repository setup or the syntax for the dependency?

LeandroSQ, 2021-11-09:
It seems that Jitpack was building the library with Java version 1.8...
I've changed to Java 11, and published v1.5.1... Still testing though

LeandroSQ, 2021-11-09:
I've created a new project to simulate your problem.

After creating it the steps I've done were:
- Add `implementation 'com.github.LeandroSQ:android-ble-made-easy:v1.5.1'` to App `build.gradle`
- Add `maven { url 'https://jitpack.io' }` inside `buildscript -> repositories` to Project `build.gradle`
- Add `maven { url 'https://jitpack.io' }` inside `dependencyResolutionManagement -> repositories` to `settings.gradle`
- Gradle sync

And done!

I'm able to use the library normally.

[image]

Swarandeep1, 2021-11-09:
Success!  It finally built and I can actually instantiate var ble : BLE  now!  I don't know if that is the correct use but I'm ready to start learning!  Thank you so much!!!

LeandroSQ, 2021-11-12:
Happy to help

closed 2021-11-12 by LeandroSQ