LeandroSQ/android-ble-made-easy #8: Using this library with Bottom Navigation

Swarandeep1, 2021-11-17:
Hello,

I've currently worked through you example and have been able to incorporate the scan and connect functionality into my app.  I am currently using a bottom Navigation app structure, where one of my bottom menu items takes the user to a fragment that allows them to set up the connection and separate bottom menu items are different fragments where data can be sent.  I've tried to setup the connection variable of type BluetoothConnection in my shared viewModel.  However, whenever I navigate to a new fragment to send data, the connection variable pulled from the shared view model is null. 

Is there an easy way to set this up using a shared view model?

Thanks!
Swarandeep Singh

Swarandeep1, 2021-11-19:
Hello,

I realized a didn't setup the connection in the new fragment.   I ended up  setting up the connection in the onCreate of the new fragments via a lifecycleScope scoped coroutine.  As I'm new to app development, I'm wondering if this will cause any unintended consequences?

Thanks, 
Swarandeep Singh

LeandroSQ, 2021-11-29:
Hello there,
You should be able to use a shared view model to store the connection. 
There must be something turning it into null.

Are you using some sort of dependency injection?

closed 2022-01-26 by LeandroSQ