LeandroSQ/android-ble-made-easy #14: Instance BLE with compose

JanissaryID, 2022-09-07:
how to use BLE with compose?

closed 2022-11-03 by LeandroSQ

LeandroSQ, 2022-11-03:
The README.md has been updated with instructions

rocketraman, 2023-06-29:
How do I get `ComponentActivity` in a Composable?

rocketraman, 2023-06-29:
It's not super-clear what to do here.

I've tried:

```
tailrec fun Context.getActivity(): AppCompatActivity? = when (this) {
  is AppCompatActivity -> this
  is Activity -> getActivity()
  is ContextWrapper -> baseContext.getActivity()
  else -> null
}

BLE(LocalContext.current.getActivity())
```

but that just results in:

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.myapp, PID: 14836
    java.lang.IllegalStateException: LifecycleOwner com.myapp.MainActivity@183df6e is attempting to register while current state is RESUMED. LifecycleOwners must call register before they are STARTED.
        at androidx.activity.result.ActivityResultRegistry.register(ActivityResultRegistry.java:123)
        at androidx.activity.ComponentActivity.registerForActivityResult(ComponentActivity.java:833)
        at androidx.activity.ComponentActivity.registerForActivityResult(ComponentActivity.java:842)
        at quevedo.soares.leandro.blemadeeasy.ContractHandler.<init>(ContractHandler.kt:21)
        at quevedo.soares.leandro.blemadeeasy.BLE.registerContracts(BLE.kt:159)
        at quevedo.soares.leandro.blemadeeasy.BLE.setup(BLE.kt:150)
        at quevedo.soares.leandro.blemadeeasy.BLE.<init>(BLE.kt:133)
```

PR #15 (2022-11-03): [Feature] Added support for ComponentActivity
- added BLE constructor for androidx.activity.ComponentActivity to support use in Jetpack Compose

closes #14