LeandroSQ/android-ble-made-easy #16: BLE Scan doesn't work with Android API Level < 29 ?

NoelChew, 2022-11-04:
Hello, thank you for creating this awesome library! 

I am currently running into some issues on older devices. I tried running the sample project with 3 different Android versions: Android 6, Android 9 and Android 10. So far only the sample app installed on the Android 10 device managed to output scan results. I have also tested the Android 6 and Android 9 devices with other BLE apps to confirm that the BLE is not faulty. Has the library been tested with pre-Android10 devices before?

LeandroSQ, 2022-11-04:
Thank you for your comment!

LeandroSQ, 2022-11-04:
I tried on Android 6 and the current default scan settings are not satisfying legacy devices.
Made a few changes on a separated branch for better defaults on legacy devices (< SDK 29)

Would you be so kind to test the sample app on [this branch](https://github.com/LeandroSQ/android-ble-made-easy/tree/feature/legacy_scan_settings) and share the results? Thank you!

NoelChew, 2022-11-05:
Tested on the following Android versions: 6, 9, 10, 11, 12.
All tests are showing expected output by displaying BLE scan results.

Thank you for the fast fix!

LeandroSQ, 2022-11-09:
No worries :)

Closed with #20

closed 2022-11-09 by LeandroSQ

LeandroSQ, 2022-12-22:
Fixed on v1.8.0