LeandroSQ/android-ble-made-easy #18: Duplicate Location Service Enabling Prompt on Android 12 Device

NoelChew, 2022-11-05:
Prompt to enable location service is shown twice on Android 12 devices. Clicking "OK" twice will cause app to crash.

Steps to reproduce crash:
1. Disable device Location Service
2. Kill app
3. Start app
4. Click "OK" to enable Location Service
5. Click "OK" to enable Location Service (again)

Logcat:
`java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=598333496, result=-1, data=Intent { (has extras) }} to activity {quevedo.soares.leandro.blemadeeasy/quevedo.soares.leandro.blemadeeasy.view.MainActivity}: java.lang.IllegalStateException: Already resumed, but proposed with update true`

NoelChew, 2022-11-05:
I found out that the ScanDeviceFragment calls onStop() when requesting to enable Bluetooth Adapter for my Android 12 device. This does not happen for my other devices. After selecting "Deny" or "Allow", the fragment will call onStart() again, which includes the call to requestPermission().

LeandroSQ, 2022-11-09:
Thank you for you finding and contribution!

I've added a comment on your PR, I'm planning on include it and the legacy device fix on the next release, and hopefully get it published by EOW.

closed 2022-12-22 by LeandroSQ

LeandroSQ, 2022-12-22:
Fixed on v1.8.0

PR #19 (2022-12-22): Update ScanDevicesFragment.kt
- moved requestPermissions() from onStart() to onViewCreated()
- closes #18