LeandroSQ/android-ble-made-easy #22: Failed to read characteristic

Octavelscx, 2022-12-13:
Hi there,

I'm pretty new with kotlin and bluetooth and I'm trying your nice library, scan device works perfectly, but when I want to connect to a device and read a characteristic I get a null return and this is what I get in log : (failed to read characteristic)

[image]

Here is my code : 

```
 * Connect BLE device
 */
@SuppressLint("MissingPermission")
public fun connectBLE(ble:BLE, device: BLEDevice) {
    var name: String = device.name.toString()
    lifecycleScope.launch {
            ble.connect(device)?.let { connection ->
                // For watching bytes
                val value = connection.read("00002A29-0000-1000-8000-00805F9B34FB")
                Log.i("BLEDATA", "$value")

        }
}
```

I read that for reading 16 bits uuid with bluetooth low energy we have make uuid this form : 
0000XXXX-0000-1000-8000–00805f9b34fb
But maybe I'm wrong ?

I got the uuid 0x180A with nrf Connect App and this is marked Read and correspond to the manufacturer brand name string. 
I bet I'm doing something wrong but dont know what...

Thanks a lot !

Octavelscx, 2022-12-14:
I think the issue comes from the BluetoothCharacteristic class

The method read uses gatt.readCharacteristic() of android library and this method is asynchronous :

https://developer.android.com/reference/android/bluetooth/BluetoothGatt

I tried with debugging and when I do breakpoints, it works and I get return value of the reading of characteristic. If I just run the app without breakpoint, it returns null value.

So I think it's about a delay of the read() method of the BluetoothCharacteristic. 

But I'm too new in kotlin to provide you an correction but I hope someone will see this and help ! :) Thanks

LeandroSQ, 2022-12-14:
Hello there 🙂
First of all, thank you for the interest. Awesome findings, these issues seem to be related to #13.

I will take a look on your suggestions as soon as possible, thanks.

LeandroSQ, 2022-12-22:
FYI, your initial guess was spot on!
I missed completely the async behavior for read();

Thank you :)

LeandroSQ, 2022-12-22:
You can check out a fix for it on #23

LeandroSQ, 2022-12-22:
Please try v1.8.0

Octavelscx, 2022-12-23:
Thanks a lot man,

I'm testing v1.8.0 right now but I get no return after the onCharacteristicRead()

This is how I call it : 

[image]

The execution never go in the if else statement... 

With the debugger I can see that it get the characteristic value : 
[image]

But the execution get stuck at this line : 

[image]

As if it never returns to my method and therefore never fits in the if else statement above

Maybe I'm calling it wrong, but I tried the other method you give and I got the same issue
For your information I attach this method to a button with a setOnClickListener like this : 

[image]

Thanks for your help

LeandroSQ, 2022-12-23:
Are you using observe too? In non-notifiable characteristics, the library fallback to a legacy observe, which uses read.
Also one limitation as per today, is only one read per characteristic at-a-time.

So when you call read you should wait until it returns to call read for the same characteristic again.

Are any of these edge scenarios affecting you?

LeandroSQ, 2022-12-24:
Would you be so kind to provide a minimum reproducible sample project?

Octavelscx, 2023-01-03:
My bad, it's perfectly working when I read notifiable characteristic, but when I want to read a non notifiable characteristic it's not working, don't really know why, but pretty sure it's on my side

Thanks again

closed 2023-01-03 by Octavelscx