LeandroSQ/android-ble-made-easy #29: scanForPeripherals cannot scan the device

panguoxian, 2023-04-05:
I modified the startScan function of BluetoothCentralManager.kt
Because the startScan function uses filter, but the filter is empty

    private fun startScan(filters: List<ScanFilter>, scanSettings: ScanSettings, scanCallback: ScanCallback) {
        if (bleNotReady()) return
        if (bluetoothScanner == null) {
            bluetoothScanner = bluetoothAdapter.bluetoothLeScanner
        }
        if (bluetoothScanner != null) {
            setScanTimer()
            currentCallback = scanCallback
            currentFilters = filters
           + if (filters.isEmpty()){
           +   bluetoothScanner?.startScan(scanCallback)
           + }else{
                bluetoothScanner?.startScan(filters, scanSettings, scanCallback)
           + }

            Logger.i(TAG, "scan started")
        } else {
            Logger.e(TAG, "starting scan failed")
        }
    }

LeandroSQ, 2023-04-05:
Hello :) 
Thank you for your issue.

I am sorry, I think I am confused, are you sure you are not referring to [blessed kotlin repo](https://github.com/weliem/blessed-android-coroutines/blob/main/blessed/src/main/java/com/welie/blessed/BluetoothCentralManager.kt#L253)? 
My library also is based on Kotlin, but has a different approach and philosophy.

panguoxian, 2023-04-06:
> 

Different methods and concepts.

I need to scan for unknown bluetooth devices nearby. How do this? i want to know your method.

I use scanForPeripherals(resultCallback: (BluetoothPeripheral, ScanResult) -> Unit, scanError: (ScanFailure) -> Unit ) function.
but resultCallback has no callback.

Because the startScan function uses filter, but the filter is empty.
So scanForPeripherals has no callback.

LeandroSQ, 2023-04-06:
You can take a look on the [README.md](https://github.com/LeandroSQ/android-ble-made-easy#installing) for this repo to see how you can install.
You can handle Bluetooth permissions, turning on the Bluetooth hardware, yourself or you can let my library help… I suggest you use the library functions to handle these, they are also on the readme file.

As for how you can scan for all nearby devices you can check [this](https://github.com/LeandroSQ/android-ble-made-easy#scanning-ble-devices). You don't have to pass filters if you don't want to, you can choose to scan for a given period or indeterminate. And you can choose to use a callback or use coroutines.

The [sample app](https://github.com/LeandroSQ/android-ble-made-easy/blob/master/app/src/main/java/quevedo/soares/leandro/blemadeeasy/view/scandevices/ScanDevicesFragment.kt) provided in this repo does the scanning for all nearby devices, you can check that for guidance.

panguoxian, 2023-04-06:
> 

Thank you for your reply

LeandroSQ, 2023-04-08:
Hey, just checking 
We're you able to scan devices successfully?

LeandroSQ, 2023-04-25:
I am closing this due to inactivity. Please let me know if you would like to re-open this issue.

closed 2023-04-25 by LeandroSQ