LeandroSQ/android-ble-made-easy #30: Default connection priority and implementing requestConnectionPriority support

Octavelscx, 2023-04-11:
Hello,

I am using your library for Bluetooth communication in my project, and I am experiencing some frame loss during data transmission. I am curious about the default connection priority used by your library and if it's possible to implement the requestConnectionPriority function to provide better control over the connection quality.

Could you please let me know the default connection priority value used in your library? Also, would it be possible to add support for the requestConnectionPriority function, so that we can adjust the connection priority based on our requirements (e.g., BluetoothGatt.CONNECTION_PRIORITY_HIGH, BluetoothGatt.CONNECTION_PRIORITY_BALANCED, or BluetoothGatt.CONNECTION_PRIORITY_LOW_POWER)? This feature would be really helpful in improving the data transmission reliability and reducing frame losses.

Thank you for your time and consideration. I appreciate your work on this library, and I am looking forward to your response.

Best regards,

LeandroSQ, 2023-04-11:
Hello there, thank you for your issue :) glad to see you again here.

That is a great idea, right now we use the `CONNECTION_PRIORITY_BALANCED` priority, we should definitely have an interface to expose this, so the user can decide which one suits best.

I will add this to my plate and will address this as soon as I have the time to do so.

LeandroSQ, 2023-04-11:
I have put out a prototype of this, would you be so kind to test this on a separate branch?

Octavelscx, 2023-04-12:
Of course but I don't know how to do that ?

LeandroSQ, 2023-04-12:
I haven't pushed the branch yet, I wanted to see if you were open to testing too.

One option is to clone the branch and test it on the Sample app this repo provides.
Another option is, I can release a version for testing, and you would simply use this version on your gradle file.

Which option do you prefer?

Octavelscx, 2023-04-12:
Ok for the branch version, it seems to be a more correct option :)

LeandroSQ, 2023-04-14:
It is up and running on the [develop](https://github.com/LeandroSQ/android-ble-made-easy/tree/develop) branch.

Let me know if that works well

Octavelscx, 2023-04-14:
Nice, is it possible to add the MTU request method added by Valexi7 because on the develop branch because I use it too and it works great? 😊

LeandroSQ, 2023-04-17:
Hey, yeah there is a PR for it, we will probably have it on another branch for now. 
My plan is to release Valexi7's changes alongside this feature, they will be on the same version.

LeandroSQ, 2023-04-26:
No worries.

FYI, we merged Valexi7's PR :)

LeandroSQ, 2023-05-20:
v1.8.2 is out, it includes this.

closed 2023-05-20 by LeandroSQ