LeandroSQ/android-ble-made-easy #31: BLE disconnected automatically after 4-5 min.

karansingla007, 2023-04-26:
When I connect with the BLE device it disconnects after 4-5 mins in Android.
But when I use the same device with windows then It does not happen.
Something is wrong.

LeandroSQ, 2023-04-26:
Hello there, thank you for your issue 🙂 

Would you be able to provide mor information a possibly a sample of how you are connecting?
Also, is the app on foreground? Are you using a background or foreground service?

karansingla007, 2023-04-26:
I am testing in Android 13. The app is in the foreground.
I am connecting like this

```
ble?.scanFor(macAddress = Constants.bleDevice.macAddress, timeout = 200000)?.let {
					onDeviceConnected(it)
					ble?.stopScan()
				}
```

```
	private fun onDeviceConnected(connection: BluetoothConnection) {
		connection.apply {
			MainActivity.connection = connection
			onConnect = {
				setDeviceConnectionStatus(true)
				setLoaderVisibility(false)
			}
			onDisconnect = {
 			 setDeviceConnectionStatus(false)
			}
			setDeviceConnectionStatus(true)
			setLoaderVisibility(false)
		}
	}
```

And after 4-5 mins onDisconnect callBack call.

LeandroSQ, 2023-04-26:
You should not need to call `ble?.stopScan` for `scanFor`.

Secondly, would you be so kind to enable verbose mode like `ble?.verbose = true` after instantiating it, and sharing the logs here?

karansingla007, 2023-04-27:
I already enabled the logs.
and i removed `ble?.stopScan`

[image]

LeandroSQ, 2023-04-28:
I don't see much there, only lost connection. Could you share these logs on a .txt file format? 

Also, I don't think this is related to your issue,  but It seems you have a ValueAnimator on a BLE callback which is not on the main thread. You can fix that by using:
```kotlin
Handler(Looper.getMainLooper()).post(Runnable {
        // things to do on the main thread
    })
```

LeandroSQ, 2023-04-28:
One thing I think it may help, I will enable more verbose logs on this disconnection scenario. And add it to a version, hopefully it will help us to catch possible causes that are actionable.

Also, in the meantime, I know it does not solve completely your problem, but could you experiment with connecting again after a disconnection? I was planning on adding this auto-connect feature, the library already supports device cache, so re-connecting should be faster.

karansingla007, 2023-05-03:
@LeandroSQ sounds good, but any estimate when can i test again?
And i tried to connect again `onDisconnect callback` but i did not receive data from bluetooth device.
I don't why.

See i am enabling a service when connection is successfully connected and then write some bytes to the bluetooth device -> device send me data in packets.

LeandroSQ, 2023-05-16:
Awesome finding!

I can put this on develop, if you are able to clone the repo and try the provided sample app to test it, then we can do a full release with this.

karansingla007, 2023-05-16:
@LeandroSQ  yeah i can test this. Lemme know when its done.

karansingla007, 2023-05-17:
@LeandroSQ can you please try this ASAP because i stuck here. i hope you will understand.
Thanks

LeandroSQ, 2023-05-17:
First of all, after looking at our source code, we do not use onReadRequest, and not commit transactions to GATT. I'm afraid this won't be helping :/

Not sure if you've seen [this stackoverflow ](https://stackoverflow.com/questions/70387604/android-12-ble-always-returning-status-0-on-disconnect) question before, and even a google issue, but this is actually a problem on Android on not showing the actual reason on why you were disconnected.
It seems we can look at the Bluetooth logs to see what is going on on your case, could you try the following and share the logs here?
[How to get the Bluetooth Host Controller Interface logs from a modern Android phone](https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa)

LeandroSQ, 2023-05-20:
Also, FYI v1.8.2 is out, which has more logging. Hopefully can help to catch your issue.

karansingla007, 2023-05-23:
I updated the version and i am getting these logs when bluetooth device disconnected.
```
D/BluetoothConnection: onCharacteristicChanged: android.bluetooth.BluetoothGattCharacteristic@438fa0b
D/BluetoothConnection: onCharacteristicChanged: android.bluetooth.BluetoothGattCharacteristic@438fa0b
D/BluetoothConnection: onCharacteristicChanged: android.bluetooth.BluetoothGattCharacteristic@438fa0b
D/BluetoothConnection: onCharacteristicChanged: android.bluetooth.BluetoothGattCharacteristic@438fa0b

D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=00:18:DA:31:89:A6
D/BluetoothConnection: onConnectionStateChange: status 0 Code: -1 - Status.Unknown - Unknown state 0 Code: 0 - State.Disconnected - Disconnected
D/BluetoothConnection: Lost connection with 00:18:DA:31:89:A6 STATUS: Code: -1 - Status.Unknown - Unknown
W/BluetoothGatt: Unhandled exception in callback
    android.util.AndroidRuntimeException: Animators may only be run on Looper threads
        at android.animation.ValueAnimator.cancel(ValueAnimator.java:1195)
        at android.view.ViewPropertyAnimator.animatePropertyBy(ViewPropertyAnimator.java:960)
        at android.view.ViewPropertyAnimator.animateProperty(ViewPropertyAnimator.java:916)
        at android.view.ViewPropertyAnimator.rotation(ViewPropertyAnimator.java:526)
        at com.aavaa.fragments.SingleDeviceFragment$onDeviceConnected$1$2.invoke(SingleDeviceFragment.kt:435)
        at com.aavaa.fragments.SingleDeviceFragment$onDeviceConnected$1$2.invoke(SingleDeviceFragment.kt:432)
        at quevedo.soares.leandro.blemadeeasy.BluetoothConnection$setupGattCallback$1.onConnectionStateChange(BluetoothConnection.kt:173)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:299)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:864)
        at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:294)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:128)
        at android.os.Binder.execTransactInternal(Binder.java:1285)
        at android.os.Binder.execTransact(Binder.java:1244)
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=00:18:DA:31:89:A6
D/BluetoothConnection: onConnectionStateChange: status 0 Code: -1 - Status.Unknown - Unknown state 2 Code: 2 - State.Connected - Connected
D/BluetoothConnection: Device 00:18:DA:31:89:A6 connected!
D/BluetoothGatt: requestConnectionPriority() - params: 0
D/BluetoothGatt: discoverServices() - device: 00:18:DA:31:89:A6
D/BluetoothGatt: onConnectionUpdated() - Device=00:18:DA:31:89:A6 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=00:18:DA:31:89:A6 Status=0
D/BluetoothConnection: onServicesDiscovered: 4 services found!
D/BluetoothGatt: onConnectionUpdated() - Device=00:18:DA:31:89:A6 interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=00:18:DA:31:89:A6 interval=9 latency=0 timeout=400 status=0
```

LeandroSQ, 2023-05-24:
One question on this last stack trace, did you reconnect before this?

karansingla007, 2023-05-24:
No, these logs when i connect very first time and it disconnected.

LeandroSQ, 2023-05-24:
Do you happen to have timestamps on this? How much time before the disconnect?

karansingla007, 2023-05-24:
Sometime after 5 min or around 2 min.

LeandroSQ, 2023-05-24:
I also found this: https://stackoverflow.com/questions/60666682/ble-connectivity-issue-when-clientif-value-is-greater-than-7

If you try restarting and turning Bluetooth ON/OFF on your device, and clientIF resets to 0. The issue persists?

karansingla007, 2023-05-24:
yes, the issue is persists in every condition.

LeandroSQ, 2023-05-26:
Were you able to get the HCI logs? They will probably have more information specific to your device, which could help us to understand what is going on for you.

karansingla007, 2023-05-29:
@LeandroSQ How can i print HCI logs?

LeandroSQ, 2023-07-27:
Closing due to inactivity, if you would like this to be re-opened, please let me know.

closed 2023-07-27 by LeandroSQ