LeandroSQ/android-ble-made-easy #38: BLE instance in service

pawelkaczmarek12, 2023-07-26:
I wanted to use the library in Foreground Service, but at the moment it's not possible. Could you please add a context-only constructor?

LeandroSQ, 2023-07-27:
Hello, Thank you for your issue
It is theoretically possible, although I have never tested this on a Service, so it would need testing.

Would you be so kind to offer help to test the library given that I expose a new constructor for you?

LeandroSQ, 2023-09-27:
Closing due to inactivity.

lemonjesus, 2023-10-18:
I'd be happy to play around with this - I have a need for BLE in a Foreground Service in a small project I'm working on and I'd love to use this library to make it happen!

LeandroSQ, 2023-10-24:
Thank you for volunteering!
I will put out a prototype with this as soon as I get time to it, so we can get this into a new version.

LeandroSQ, 2023-10-25:
Just updating this with my findings.

1. We need [special permission](https://developer.android.com/reference/android/Manifest.permission#FOREGROUND_SERVICE_CONNECTED_DEVICE) to handle connected devices on a foreground service 
2. Couldn't find a way to handle the jetpack contracts on the services, so it may be possible that using helpers like hardware availability, permission checking and so on wouldn't work
3. It may be only possible to start the BLE instance on an Activity, Fragment or Jetpack compose activity, connect to a device and then pass the connection instance to the service.