LeandroSQ/android-ble-made-easy #39: Characteristic not found on device when using the sample app

sarimmehdi, 2023-08-24:
We have a custom Bluetooth module we want to read and write to. I am able to connect to it but when I tap on TOGGLE, I get **"Current status: Information not sent"**. Investigating this revealed that the getCharacteristic method in BluetoothConnection returns null right after this message is logged: **Characteristic 4ac8a682-9736-4e5d-932b-e9b31405049c not found on device C6:2F:1D:80:A7:8A!**

**4ac8a682-9736-4e5d-932b-e9b31405049c** is the characteristic in the sample app. How can I find the right characteristic to use with my Bluetooth device?

LeandroSQ, 2023-08-24:
Hello there, thank you for your interest in the library.

Different devices will expose different characteristics, the one on the sample app is just an example.

If the device you have exposed the characteristics you can fetch them by using the library.

After connecting to your device, the connection instance will have the following properties:
- notifiableCharacteristics
- readableCharacteristics
- writableCharacteristics

You can use them to list the available and exposed characteristics from your device. If they're empty that means nothing is being exposed, so you are better of trying to contact the manufacturer.

LeandroSQ, 2023-08-30:
Did it work?

LeandroSQ, 2023-09-27:
Closing due to inactivity.

closed 2023-09-27 by LeandroSQ