LeandroSQ/android-ble-made-easy #42: MTU value not being updated

Marco012, 2023-10-19:
Hello, 

The requestMTU(bytes) method is not requesting the MTU with the bytes we are sending, but instead is just requesting with its current MTU value. I was having issues with this, and changing the "mtu" to "bytes" in the following line fixed it.

https://github.com/LeandroSQ/android-ble-made-easy/blob/b483665e044e19c67fb4ce9e79381f040cb1f0d2/lib/src/main/java/quevedo/soares/leandro/blemadeeasy/BluetoothConnection.kt#L311C26-L311C26

LeandroSQ, 2023-10-24:
First of all, thank you for your issue and interest on the library.
Secondly, great catch!

Do you want to open a PR for this? So your name would appear as a contributor :)

Marco012, 2023-10-24:
The library is great!

I opened a PR with this fix.

Continue with your great work 😁

LeandroSQ, 2023-10-24:
I appreciate your words! Thank you

And welcome as a contributor, your name is already showing up as an official contributor to this project.

closed 2023-10-24 by LeandroSQ