LeandroSQ/android-ble-made-easy #48: 2 files found with path 'META-INF/NOTICE.md'

black, 2024-09-23:
I have just this library to build in my blank Android project and I get the following error upon building 

```
2 files found with path 'META-INF/NOTICE.md'.
Adding a packaging block may help, please refer to
https://developer.android.com/reference/tools/gradle-api/8.5/com/android/build/api/dsl/Packaging
for more information
```

My Gradle file  

  
    namespace = "com.experiment.bletest"
    compileSdk = 34

    defaultConfig {
        applicationId = "com.experiment.bletest"
        minSdk = 24
        targetSdk = 34
        versionCode = 1
        versionName = "1.0"

        testInstrumentationRunner = "androidx.test.runner.AndroidJUnitRunner"
    }

LeandroSQ, 2024-09-23:
Hello, thank you for opening the issue

Would you be so kind to share the whole project? So I can take a deeper look

Thank you

black, 2024-09-23:
Hey,
I just created a blank project and added your library and try to build the project. I haven't added anything else. I am using latest Android Studio.

LeandroSQ, 2024-09-24:
I changed the packaging exclusion on be282decc023225997ead4086d072e75572a2979
Please try 1.9.3 as let me know if it works for you :)

black, 2024-09-24:
awesome thank you. Let me try that.

black, 2024-10-07:
Hey Sorry, I was a bit busy.  I checked your updated library and it wasn't working. I had to add packaging exclusion in order to work.