LeandroSQ/android-ble-made-easy #49: How do I read multiple characteristics ?

black, 2024-10-10:
I am using the below code to read the multiple characteristics but only the first characteristics gets observed .  I have remove the first observer then I started receiving from the next and so on. Basically it is only reading one characteristics at a time. 

```
ble.connect(device)?.let { connection ->
                connection.requestMTU(bytes = 64)
                binding.connectionStatus.text = if(connection.isActive) "Connected" else "not Connected"
                connection.apply {
                    observeString(characteristic = CHARACTERISTIC_UUID_ECG, charset = Charsets.UTF_8) { value: String ->
                        Log.d(TAG, "ECG::: $value")
                        binding.dataEcg.text = "ECG::: $value"
                    }
                    observeString(characteristic = CHARACTERISTIC_UUID_IMU, charset = Charsets.UTF_8) { value: String ->
                        Log.d(TAG, "IMU::: $value")
                        binding.dataImu.text = "IMU::: $value"
                    }
                    observeString(characteristic = CHARACTERISTIC_UUID_BATTERY, charset = Charsets.UTF_8) { value: String ->
                        Log.d(TAG, "BAT::: $value")
                        binding.dataBattery.text = "BAT::: $value"
                    }
                }

            }
```

soheilkooklan, 2024-10-12:
Hi There!
The issue is that only one characteristic is being observed at a time because multiple observers are set up for conflicting traits. 
To fix this, try setting up each observer sequentially with a slight delay between each one to give them time to establish properly. Here's a suggestion to modify your code using coroutines and `launch` to handle multiple observers without interference:
- **Coroutines (`GlobalScope.launch`)**: This ensures each characteristic gets its independent observer without blocking or interfering with the others.
- **Separation**: Each characteristic now observes its data in a non-blocking way, ensuring that all the observers can function simultaneously.

Give this a try, and it should help you read from multiple characteristics at the same time!

```
ble.connect(device)?.let { connection ->
    connection.requestMTU(bytes = 64)
    binding.connectionStatus.text = if (connection.isActive) "Connected" else "Not Connected"

    connection.apply {
        // Launch a coroutine to observe each characteristic sequentially
        GlobalScope.launch {
            observeString(characteristic = CHARACTERISTIC_UUID_ECG, charset = Charsets.UTF_8) { value: String ->
                Log.d(TAG, "ECG::: $value")
                binding.dataEcg.text = "ECG::: $value"
            }
        }

        GlobalScope.launch {
            observeString(characteristic = CHARACTERISTIC_UUID_IMU, charset = Charsets.UTF_8) { value: String ->
                Log.d(TAG, "IMU::: $value")
                binding.dataImu.text = "IMU::: $value"
            }
        }

        GlobalScope.launch {
            observeString(characteristic = CHARACTERISTIC_UUID_BATTERY, charset = Charsets.UTF_8) { value: String ->
                Log.d(TAG, "BAT::: $value")
                binding.dataBattery.text = "BAT::: $value"
            }
        }
    }
}
```

black, 2024-10-14:
@soheilkooklan I tried this already. This was the first response from claudeAi and ChatGPT both. It didn't work.

LeandroSQ, 2024-10-14:
Hello @black thanks for the interest in the library :)

black, 2024-10-14:
@LeandroSQ Hey, Can you help me with this one please?

black, 2024-10-14:
Yes, they do.

LeandroSQ, 2024-10-14:
And all of these read correctly if done one-by-one?

LeandroSQ, 2024-10-14:
Another thing, are you able to put a breakpoint [here](https://github.com/LeandroSQ/android-ble-made-easy/blob/dfe096d08cec36a6f17271633e0197b9ef213f52/lib/src/main/java/quevedo/soares/leandro/blemadeeasy/BluetoothConnection.kt#L487) and check if this gets called?

black, 2024-10-14:
> And all of these read correctly if done one-by-one?

Yes, that correct.

black, 2024-10-14:
> Another thing, are you able to put a breakpoint [here](https://github.com/LeandroSQ/android-ble-made-easy/blob/dfe096d08cec36a6f17271633e0197b9ef213f52/lib/src/main/java/quevedo/soares/leandro/blemadeeasy/BluetoothConnection.kt#L487) and check if this gets called?

Hey This isn't getting called.

soheilkooklan, 2024-10-14:
> @soheilkooklan I tried this already. This was the first response from claudeAi and ChatGPT both. It didn't work.

**That code was rewritten with ChatGPT. Another way is to check these options:**
- Although you requested an MTU size 64, some devices may still have restrictions.
- There could be limitations in the Bluetooth library or API you're using that may prevent simultaneous observation.

black, 2024-10-15:
@soheilkooklan Pleases stop posting ChatGPT response. If I want, I could take help from ChatGPT myself.  Stop spamming.

soheilkooklan, 2024-10-15:
> @soheilkooklan Pleases stop posting ChatGPT response. If I want, I could take help from ChatGPT myself. Stop spamming.
Your device may not be working with the features you want. I rewrote the code with ChatGPT just to save time and help you. There's no need to be rude—everyone here has problems they're trying to solve, just like you. This isn’t spam; it’s an attempt to offer assistance.

LeandroSQ, 2024-10-15:
> Another thing, are you able to put a breakpoint [here](https://github.com/LeandroSQ/android-ble-made-easy/blob/dfe096d08cec36a6f17271633e0197b9ef213f52/lib/src/main/java/quevedo/soares/leandro/blemadeeasy/BluetoothConnection.kt#L487) and check if this gets called?

Were you able to test this? @black 

Also @soheilkooklan, I understand it might be a very specific issue with either the library or device, not sure if LLMs can help going forward. But let me know.

LeandroSQ, 2024-10-15:
Sorry I missed your comment.

So if the breakpoint is not being called it is most likely that your device is not receiving updates.

Is there any other device you can try your app? Or even any other emitter device you can connect to?

It may be a limitation, as per here https://stackoverflow.com/a/42126184

Will do some testing later as well and will let you know.

black, 2024-10-16:
@LeandroSQ So I have tried three other apps to test my ble stream from Seeed xiao NRF 54320 board - LightBlue, nrf connect and Serial Bluetooth. I am getting my characteristics read simultaneously.  

Let me check that stackoverflow and get back to you.

black, 2024-10-17:
@LeandroSQ I am able to fix it. Basically I needed to add delay between the observers. 

```
          ble.connect(device)?.let { connection ->
              connection.requestMTU(bytes = 64)
              binding.connectionStatus.text = if(connection.isActive) "Connected" else "not Connected"
              connection.apply {
                  observeString(characteristic = CHARACTERISTIC_UUID_ECG, charset = Charsets.UTF_8) { value: String ->
                      Log.d(TAG, "ECG::: $value")
                      binding.dataEcg.text = "ECG::: $value"
                  }

                  delay(1000) // <----- adding delay fixed the code and started receiving the data on all the characecteristics

                  observeString(characteristic = CHARACTERISTIC_UUID_IMU, charset = Charsets.UTF_8) { value: String ->
                      Log.d(TAG, "IMU::: $value")
                      binding.dataImu.text = "IMU::: $value"
                  }

                  delay(1000) // <----- adding delay fixed the code and started receiving the data on all the characecteristics

                  observeString(characteristic = CHARACTERISTIC_UUID_BATTERY, charset = Charsets.UTF_8) { value: String ->
                      Log.d(TAG, "BAT::: $value")
                      binding.dataBattery.text = "BAT::: $value"
                  }
              }

          }
```

closed 2024-10-17 by black

LeandroSQ, 2024-10-17:
I see, we probably will have to queue these operations then.

Thank you for finding that! I really appreciate it.